actuallymentor/battery #222: limit hysteresis would be helpful

Kaiman77, 2024-01-15:
**Is your feature request related to a problem? Please describe.**
When the battery reachd the 80% limit, everything is fine until it drops to 79% for whatever reason.
Then the tool triggers the charging up to 80% again until the battery drops to 79% again.

**Describe the solution you'd like**
Adding the possibility for a hysteresis would be helpful.
Basically saying, charge up to 80% only if the battery drops below 70% (or whatever value).

**Describe alternatives you've considered**
I don't see an alternative, that's a pretty typical way to do it in automation control

actuallymentor, 2024-01-17:
Cool idea. Do you have the time/bandwidth to draft a PR?

Kaiman77, 2024-01-17:
time/bandwidth yes, but my skill level doesn't allow me to, sorry ;-)

actuallymentor, 2024-01-17:
Alas, I will tag it as a feature request and hope that someone with more time than me can pick it up :)

earthsound, 2024-01-17:
@actuallymentor dupe of issue #150?

seefood, 2024-02-12:
Yup, seems to me like a clone of #150, which includes some decent bash and applescript workarounds. @Kaiman77 I suggest closing.

Kaiman77, 2024-02-18:
I agree to close as duplicate of #150