actuallymentor/battery #223: Sleeping momentarily while charging causes issues

ballo, 2024-01-17:
**What is the issue? (required)**
Wacky unwanted behavior if the mac sleeps when battery is running. 

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. run battery discharge 40 when battery is a little below 40
2. sleep for a second or so while battery is running (how you know the macbook is sleeping requires mystic powers not covered here. Apple likes to keep this a secret, shhh!)
3. wake mac
4. mac will completely charge battery to 100% (no idea why)
5. if the mac is put to sleep after battery is no longer running, the mac will behave like the charger cable is being constantly unplugged then replugged (charge beep, no charge, repeat) 
6. The saga continues? Will I have to reboot or SMC reset? Will I remember to run battery after the reboot to avoid 100% charging? Stay tuned I guess. For now I'll keep it unplugged because the battery is overcharged. 

**Expected behavior (required)**
I was kinda hoping battery discharge 40 would charge the battery to 40%

actuallymentor, 2024-01-17:
I've seen sporadic reports of this but can't reproduce. Do you have a reliable way to trigger this issue so I can isolate it?

ballo, 2024-01-17:
It's pretty difficult to even know if an M1 macbook is even sleeping.

ballo, 2024-01-18:
and now it's refusing to charge. Last time this happened I had to reboot