actuallymentor/battery #225: What changes should I make to the "Optimized Battery Charging" setting?

seamusdemora, 2024-01-18:
**What is the issue? (required)**
This isn't a "bug" per se - or maybe a bug in the documentation - or maybe I've just been unable to find it??

**What exactly did you do to produce the issue? (required)**
I've not actually installed the app yet; this is a "pre-install" Question

**Expected behavior (required)**
I would like to know how to set my "Optimized Battery Charging" option to allow `battery` to operate effectively. I am **guessing** that it should be turned OFF/Disabled, but cannot find anything in the docs to confirm that. 

Maybe this should be addressed in the README.md file???

**Screenshots (optional)**

**"Optimized Battery Charging" - ENABLED --OR-- DISABLED ?**

[image]

**Error logs**
N/A

Did `battery` fail to install? Use `tail -n 1000 ~/.battery/*/log` instead.
N/A

**Additional context (optional)**
Add any other context about the problem here.

cjosue15, 2024-01-20:
I've been looking for that in the documentation, but haven't found anything.

At least in Aldente it is necessary to disable the "optimized loader" for the application to work properly.

seamusdemora, 2024-02-06:
Aw, c'mon... this should be a simple question for anyone associated with development of this project... It's been 3 weeks now, and all I hear is "crickets".

aschiletus, 2024-03-28:
That is the underlying idea. We are limiting with this application, we don't need anything different.

seamusdemora, 2024-03-28:
> That is the underlying idea. We are limiting with this application, we don't need anything different.

?? What does that mean... could you please put that in plain English?

aschiletus, 2024-03-28:
> @aschiletus
> 
> > That is the underlying idea. We are limiting with this application, we don't need anything different.
> 
> ?? What does that mean... could you please put that in plain English?

What changes should I make to the "Optimized Battery Charging" setting?

You should keep it off.

seamusdemora, 2024-03-28:
You know...  this project is a waste of time and effort. It's going nowhere, but apparently no one has grasped that as yet.

Outta' here!

aschiletus, 2024-03-30:
> You know... this project is a waste of time and effort. It's going nowhere, but apparently no one has grasped that as yet.
> 
> Outta' here!

Sorry for my bad English. I think we misunderstood something. The purpose of this project is simple and it does it in the best way. There is no need for constant updates because its purpose is clear and simple.

54huanan, 2024-04-17:
There is no need to enable Optimized Battery Charging anymore, as this app can do better.