actuallymentor/battery #234: Time Machine backups not starting reliably when "Back up on battery power" is off

blackketter, 2024-02-14:
Not sure if this is an actual bug, documentation suggestion, or what, but I've found that my automatic Time Machine backups have not been kicking in reliably.  I wonder if it's because the TM setting "Back up on battery power" has been turned off.  

It appears that if Time Machine tries to start a backup and Battery has paused charging, then the backup doesn't begin (and an attempt is recorded in the Time Machine preferences plist).  

Turning on "Back up on battery power" seems to make it more reliable.

winstonford, 2024-02-16:
I am backing up every day on auto with Backup on battery power OFF, and so far no problems.  I back up to a 1TB sd card that remains in the slot.

blackketter, 2024-02-19:
My Time Machine issues don't appear to be related to Battery.  Sorry for the false alarm.

closed 2024-02-19 by blackketter