actuallymentor/battery #235: My Mag Safe does not work anymore until I installed Battery

l0owl0ow, 2024-02-19:
My Mag Safe turned orange blinking until I installed battery.

1. I installed the DMG
2. my MacBook M3 mag safe turned Orange blinking from then

Is there a way I can uninstall completely Battery ? I tried to trash but it does not work. Even after uninstalling my battery sticks to 80%

[image]

Installation were great but now I cannot use my magsafe anymore. Apple propose me to replace my computer to fix that. Is there any way to not change my computer ?

Thousand of thanks 😊

l0owl0ow, 2024-02-20:
@actuallymentor Can you please help 🙏🏻🙏🏻🙏🏻🙏🏻🙏🏻

thdrmrphy, 2024-02-22:
Have you tried fixing it with the CLI? Open Terminal and type:
```
battery charging on
```
Hit enter.
This should allow you to charge it again. Maybe try restarting. 
If that fails, remove the tool from your computer:
```
battery uninstall
```

Hope this helps!