actuallymentor/battery #245: Menu Bar icons

42648594, 2024-03-07:
**Is your feature request related to a problem? Please describe.**
A couple different battery icons as well as a charge percentge display, such that we can replace the native battery icon in the Menu Bar without losing any of the information it shows.

**Describe the solution you'd like**
A small selection of Menu Bar icons, akin to iStat Menu battery level icons, etc.

**Describe alternatives you've considered**
Rotating the battery 90º clockwise would make it almost perfectly replicate the native battery icon, for example.

bearwithtea, 2024-04-11:
bumping this because I love this project, but the icon being different than the MacOS icon is slightly annoying

hthomee, 2024-05-08:
I, too love this project and would love to see this change to seemlessly replace the native battery icon. I'm happy to cover the design part and provide the icons.