actuallymentor/battery #247: recurring message on every reboot.

Gainsbar77, 2024-03-13:
Hello,
Every day I get this message:
After reinstallation, it works again. I don't understand why I get this message.
« Battery needs to apply a backwards incompatible update, to do this it will ask for your password. This should not happen frequently. »
I have a macbook pro M2 pro 14 chips from march 2023.
Thanks if you have any information to stabilize the software.
I have of course destinaller and reinstaller the product but it doesn't change anything.
But thank you, the software works very well.
Thanks in advance.

NapoelonIV, 2024-07-08:
This happens every time the app is relaunched. There is another bug report about this.

rodalpho, 2024-10-07:
Was this ever fixed? It is extremely annoying, as touchID doesn't work to authenticate.