actuallymentor/battery #249: Battery maintain level does not work when mac is closed and sleeping

R4gn4r-spec, 2024-03-27:
2023 macbook pro m2 max 

When macbook lid is closed and sleeping (not in clamshell mode) the battery maintain function does not work. I have the battery maintain level set to 80%. This works great in clamshell mode and when mac is open and awake. However if macbook is closed the battery charges to 100% every time.

Gainsbar77, 2024-03-28:
Hello,

I'd like to give my opinion.
I'm not shocked because it's an application. So if the system isn't open, the application can't do its job.
Personally, I never leave my Macbook charging when it's switched off.

Benjoyo, 2024-04-02:
> Hello,
> 
> I'd like to give my opinion. I'm not shocked because it's an application. So if the system isn't open, the application can't do its job. Personally, I never leave my Macbook charging when it's switched off.

It adjusts a setting in smc which should be persistent however.

I am using the tool for a few days and it seemed to work, but then yesterday I think I closed the lid and it charged to 90% (not 100). Strange.

Gainsbar77, 2024-04-04:
Hello,

It's actually quite strange, because I don't have the same behaviour on my macbook pro M1. Whether it's switched off or in standby with the cover closed, if I leave the charger on I get up to 100%.

jannek, 2024-04-26:
Same behaviour here, Macbook Pro M3 Pro. If I can provide some logs to assist in debug, let me know.

pburner, 2024-04-27:
Is this going to be fixed?

Crousti37, 2024-05-09:
Same issue here on a macbook M2 pro. Battery app installed via .dmg file.