actuallymentor/battery #252: 【help】install error：Battery limiter error：

justincnn, 2024-04-03:
Hello, I am encountering an error while installing version 1.2.1. The error message is as follows:

Please help, thank you.

[image]

andrewhzy, 2024-04-04:
I got this too

masakiaota, 2024-04-05:
I got this too. See https://github.com/actuallymentor/battery/issues/182

krzysztofjeziorny, 2024-04-13:
I've got a similar one, installing with both `brew` and from release's dmg:

[image]

MacBook Pro 14, M3 Pro, Sonoma 14.4.1

goldjunge91, 2024-04-17:
> I've got a similar one, installing with both `brew` and from release's dmg:
> 
> <img alt="Screenshot 2024-04-13 at 13 50 30" width="260" src="https://private-user-images.githubusercontent.com/872730/322203895-99fcb987-92cc-4630-9237-5d45cd9a9d9b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzNzMwNjYsIm5iZiI6MTcxMzM3Mjc2NiwicGF0aCI6Ii84NzI3MzAvMzIyMjAzODk1LTk5ZmNiOTg3LTkyY2MtNDYzMC05MjM3LTVkNDVjZDlhOWQ5Yi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxN1QxNjUyNDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04MDkzZjY5Njc5YmRhOGFlNzIxNjBiMzIwNDcwOTBmNTIyM2NjN2E2YWY1NzRkZTJmZjNhMjRhODQyMDQ1MWIyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.yLdHE9OiRqSEmuFrhPpNXLHc-Tqo3qxhiXbM2UBr4m0">
> MacBook Pro 14, M3 Pro, Sonoma 14.4.1

I get the same error. 
Macbook Air M1 14.4.1 (23E224)

I try to Reitstall nothing Works.

fexxdev, 2024-04-20:
> I've got a similar one, installing with both `brew` and from release's dmg:
>
> <img alt="Screenshot 2024-04-13 at 13 50 30" width="260" [...]

Same. I dont know what to do really... tried uninstaling and reinstalling with brew, no luck.