actuallymentor/battery #273: Battery app pops up a GUI alert message stating "Battery limiter error: Cannot read properties of undefined (reading 'includes')"

johnmcdowell, 2024-05-13:
**What is the issue? (required)**
Battery app pops up a GUI alert message stating "Battery limiter error: Cannot read properties of undefined (reading 'includes')"

Further, the app does not log the output of its call to `battery status_csv` when that call does not return expected output.

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Let computer hibernate/deep sleep from low battery.
2. Plug in computer and log in
3. Observe GUI error stating "Battery limiter error: Cannot read properties of undefined (reading 'includes')" and logs below.

I'm not sure I can repro this consistently, but the app normally works and the error is intermittent.

**Expected behavior (required)**
Not see a GUI error message at log-in.

**Error logs**
```
[ update-electron-app ] 
checking-for-update
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew/bin:/opt/homebrew/sbin:/opt/homebrew battery status_csv
Battery status: 
{"percentage":"??","remaining":"unknown","charging":false,"discharging":false,"battery_state":"??% (unknown remaining)","daemon_state":"smc charging disabled"}
Setting interface refresh speed to 0.5 minutes
Battery status: 
{"percentage":"??","remaining":"unknown","charging":false,"discharging":false,"battery_state":"??% (unknown remaining)","daemon_state":"smc charging disabled"}
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew/bin:/opt/homebrew/sbin:/opt/homebrew battery status
[ update-electron-app ] 
updater error
[ update-electron-app ] 
Error: The Internet connection appears to be offline.
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew/bin:/opt/homebrew/sbin:/opt/homebrew battery status_csv
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew/bin:/opt/homebrew/sbin:/opt/homebrew battery status_csv
Generate app menu percentage: ?? (discharge disallowed, limited off)
Error getting battery status: 
TypeError: Cannot read properties of undefined (reading 'includes')
Display percentage NaN based on NaN
Get active logo for ??
🚨 Missing image: /Applications/battery.app/Contents/Resources/battery-inactive-NaN-Template.png
Refreshing interface update timer
Resetting interface timer speed
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew/bin:/opt/homebrew/sbin:/opt/homebrew battery status_csv
Refreshing tray icon...
Battery status: 
{"percentage":"??","remaining":"unknown","charging":false,"discharging":false,"battery_state":"??% (unknown remaining)","daemon_state":"smc charging disabled"}
Setting interface refresh speed to 0.5 minutes
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew/bin:/opt/homebrew/sbin:/opt/homebrew battery status_csv
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew/bin:/opt/homebrew/sbin:/opt/homebrew battery status
Battery status: 
{"percentage":"??","remaining":"unknown","charging":false,"discharging":false,"battery_state":"??% (unknown remaining)","daemon_state":"smc charging disabled"}
Limiter status message: 

Refreshing interface update timer
Refreshing tray icon...
Force discharge setting: undefined undefined
[ update-electron-app ] 
checking-for-update
Battery status: 
{"percentage":"21","remaining":"2:01","charging":true,"discharging":false,"maintain_percentage":"80","battery_state":"21% (2:01 remaining)","daemon_state":"smc charging enabled"}
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew/bin:/opt/homebrew/sbin:/opt/homebrew battery status
Battery status: 
{"percentage":"21","remaining":"2:01","charging":true,"discharging":false,"maintain_percentage":"80","battery_state":"21% (2:01 remaining)","daemon_state":"smc charging enabled"}
Setting interface refresh speed to 0.5 minutes
Limiter status message: 
05/13/24-11:30:06 - Battery at 21% (2:01 remaining), smc charging enabled
05/13/24-11:30:06 - Your battery is currently being maintained at 80%

Get active logo for ??
Generate app menu percentage: ?? (discharge disallowed, limited on)
🚨 Missing image: /Applications/battery.app/Contents/Resources/battery-active-NaN-Template.png
```

johnmcdowell, 2024-05-13:
I tried to debug the issue with the logs.

I think there is a case here where exec_async can resolve a promise with and undefined value if `battery status_csv` prints nothing to stdout? Maybe there are other similar cases here as well?
```
const is_limiter_enabled = async () => {

    try {
        const message = await exec_async( `${ battery } status` )
        log( `Limiter status message: `, message )
        return message.includes( 'being maintained at' )
```

I also wonder if there shouldn't be a log entry in the case that message is undefined here, rather than just silently assigning empty values? Though perhaps that doesn't help us understand the root of the issue.
```
const get_battery_status = async () => {

    try {
        const message = await exec_async( `${ battery } status_csv` )
        let [ percentage='??', remaining='', charging='', discharging='', maintain_percentage='' ] = message?.split( ',' ) || []
```

All that said, I don't understand if I'm seeing the popup alert after hibernation because of something that needs to be fixed deeper down, or if instead that GUI alert should suppressed in some cases.

Hope this helps.

jhaemin, 2024-09-07:
It happens sporadically, any updates?

SuperArtus, 2024-09-20:
Hello, same problem for me I post an attachment with the popup error. The popup appear when my MacBook is plug on my charger but I stop my power bank. I supposed the value is like "Plug on power but no power delivery" I don't know if it's understandable.
[image]

spekulazarus, 2025-03-09:
same for me