actuallymentor/battery #276: Install just the CLI through brew

FezVrasta, 2024-05-19:
Could you add just the CLI installer to Brew?

RyanBalfanz, 2024-06-05:
> Could you add just the CLI installer to Brew?

https://github.com/actuallymentor/battery?tab=readme-ov-file#-command-line-version

FezVrasta, 2024-06-10:
> > Could you add just the CLI installer to Brew?
> 
> It is not via Homebrew but did you see this? [actuallymentor/battery#-command-line-version](https://github.com/actuallymentor/battery?tab=readme-ov-file#-command-line-version)

Yes, I'm asking for Brew support though.

RyanBalfanz, 2024-06-10:
> Yes, I'm asking for Brew support though.

I don't have any say in this, but I'm wondering if you're asking because you don't want the menu item. If instead of a separate brew formula installable, would an option to not show the dock or menubar item satisfy what you want? Perhaps that setting could also be controlled via the CLI.

FezVrasta, 2024-06-11:
No I just want the CLI, I don't want an electron app, either running in background or on the menu bar.