actuallymentor/battery #278: App is Broken, do NOT INSTALL THIS

993Driver, 2024-05-24:
**What is the issue? (required)**
A clear and concise description of what the bug is.

GUI does not change charging behavior, in fact it does nearly nothing. The only features that work are enabling the 80% limit and force discharging to the 80% limit. For everything else you must rely on terminal command lines. If you uninstall there are a ton of errors on reboot, and you'll be stuck with whatever battery % was setup in command line, or 80% in GUI.

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Go to Battery GUI
2. Click on 'Disable or Uninstall'....'
3. Scroll down to 'Disable or Uninstall....'
4. See error

**Expected behavior (required)**
Expect the disable to actually stop the 80% limit. Expect uninstall to remove entirely and restore back to normal charging function. Does not. 

**Screenshots (optional)**

[image]

[image]

**Error logs**
If you know how a terminal works, please paste the content of `battery logs` here. This allows me to see potential errors on your device.

👾 Battery CLI logs:

05/22/24-09:08:22 - Battery at 95% (target 80%)
05/22/24-09:09:22 - Battery at 95% (target 80%)
05/22/24-09:10:22 - Battery at 95% (target 80%)
05/22/24-09:11:22 - Battery at 95% (target 80%)
05/22/24-09:12:22 - Battery at 94% (target 80%)
05/22/24-09:13:22 - Battery at 94% (target 80%)
05/22/24-09:14:22 - Battery at 94% (target 80%)
05/22/24-09:15:22 - Battery at 94% (target 80%)
05/22/24-09:16:22 - Battery at 94% (target 80%)
05/22/24-09:17:22 - Battery at 93% (target 80%)
05/22/24-09:18:22 - Battery at 93% (target 80%)
05/22/24-09:19:22 - Battery at 93% (target 80%)
05/22/24-09:20:23 - Battery at 93% (target 80%)
05/22/24-09:21:23 - Battery at 93% (target 80%)
05/22/24-09:22:23 - Battery at 93% (target 80%)
05/22/24-09:23:23 - Battery at 92% (target 80%)
05/22/24-09:24:23 - Battery at 92% (target 80%)
05/22/24-09:25:23 - Battery at 92% (target 80%)
05/22/24-09:26:23 - Battery at 92% (target 80%)
05/22/24-09:27:23 - Battery at 91% (target 80%)
05/22/24-09:28:23 - Battery at 91% (target 80%)
05/22/24-09:29:23 - Battery at 91% (target 80%)
05/22/24-09:30:23 - Battery at 90% (target 80%)
05/22/24-09:31:23 - Battery at 90% (target 80%)
05/22/24-09:32:23 - Battery at 90% (target 80%)
05/22/24-09:33:23 - Battery at 90% (target 80%)
05/22/24-09:34:23 - Battery at 90% (target 80%)
05/22/24-09:35:23 - Battery at 89% (target 80%)
05/22/24-09:42:01 - Battery at 89% (target 80%)
05/22/24-09:43:01 - Battery at 89% (target 80%)
05/22/24-09:44:01 - Battery at 89% (target 80%)
05/22/24-09:45:01 - Battery at 89% (target 80%)
05/22/24-09:46:01 - Battery at 89% (target 80%)
05/22/24-09:47:01 - Battery at 88% (target 80%)
05/22/24-09:48:01 - Battery at 88% (target 80%)
05/22/24-09:49:01 - Battery at 88% (target 80%)
05/22/24-09:50:02 - Battery at 88% (target 80%)
05/22/24-09:51:02 - Battery at 88% (target 80%)
05/22/24-09:52:02 - Battery at 88% (target 80%)
05/22/24-09:53:02 - Battery at 87% (target 80%)
05/22/24-10:06:41 - Battery at 87% (target 80%)
05/22/24-10:07:41 - Battery at 87% (target 80%)
05/22/24-10:08:41 - Battery at 87% (target 80%)
05/22/24-10:09:41 - Battery at 87% (target 80%)
[... 155 lines cut ...]
Display percentage 80 based on 80
Battery status: 
{"percentage":"80","remaining":"unknown","charging":false,"discharging":false,"maintain_percentage":"80","battery_state":"80% (unknown remaining)","daemon_state":"smc charging disabled"}
Setting interface refresh speed to 10 minutes

📁 Config folder details:

total 680
drwxr-xr-x@  6 brett  staff   192B May 23 14:56 .
drwxr-x---+ 16 brett  staff   512B May 23 14:56 ..
-rwxr-xr-x@  1 brett  staff   6.5K May 24 08:38 battery.log
-rw-r--r--@  1 brett  staff   268K May 24 08:34 gui.log
-rw-r--r--@  1 brett  staff     3B May 23 11:59 maintain.percentage
-rw-r--r--   1 root   staff   897B May 23 14:56 visudo.tmp

⚙️  Battery data:

05/24/24-08:38:00 - Battery at 80% (attached; remaining), smc charging disabled
Battery CLI utility v1.1.6
brett@bretts-mbp ~ % 

Did `battery` fail to install? Use `tail -n 1000 ~/.battery/*/log` instead.

**Additional context (optional)**
Add any other context about the problem here.

blackradysh, 2024-06-02:
Hi,
I confirm that the app is broken. I don't know if this is related to the recent macOS updates.

I managed to restore my Mac's charging behavior with the following 2 steps:
1. By uninstalling the application in CLI with the `battery uninstall` command.
2 By resetting my SMC using the [Apple procedure](https://support.apple.com/en-us/102605)

fdmsh, 2024-08-20:
> EDIT: I'm going to go try this: https://github.com/zackelia/bclm 

Is it any better?