actuallymentor/battery #280: Icon not showing when <=10% battery

jakjakob, 2024-05-28:
Pretty self explanatory.

[image]

<details><summary>Logs</summary>
<p>

👾 Battery CLI logs:

05/07/24-10:53:54 - 🔌🪫 Disabling battery charging
05/07/24-10:53:54 - 💡 Setting magsafe color to green
05/07/24-13:31:20 - Charge below 80
05/07/24-13:31:20 - 🔌🔋 Enabling battery charging
05/07/24-13:31:20 - 🔼🪫 Disabling battery discharging
05/07/24-13:31:21 - 💡 Setting magsafe color to orange
05/09/24-20:19:35 - Not triggering discharge as it is not requested
05/09/24-20:19:35 - Charging to and maintaining at 80% from 14%
05/09/24-20:19:35 - Charge below 80
05/09/24-20:19:35 - 🔌🔋 Enabling battery charging
05/09/24-20:19:35 - 🔼🪫 Disabling battery discharging
05/09/24-20:19:35 - 💡 Setting magsafe color to orange
05/09/24-22:46:58 - Not triggering discharge as it is not requested
05/09/24-22:46:58 - Charging to and maintaining at 80% from 26%
05/09/24-22:46:58 - Charge below 80
05/09/24-22:46:58 - 🔌🔋 Enabling battery charging
05/09/24-22:46:58 - 🔼🪫 Disabling battery discharging
05/09/24-22:46:58 - 💡 Setting magsafe color to orange
05/10/24-20:55:54 - Not triggering discharge as it is not requested
05/10/24-20:55:54 - Charging to and maintaining at 80% from 15%
05/10/24-20:55:54 - Charge below 80
05/10/24-20:55:54 - 🔌🔋 Enabling battery charging
05/10/24-20:55:54 - 🔼🪫 Disabling battery discharging
05/10/24-20:55:54 - 💡 Setting magsafe color to orange
05/10/24-21:39:56 - Charge above 80
05/10/24-21:39:56 - 🔌🪫 Disabling battery charging
05/10/24-21:39:56 - 💡 Setting magsafe color to green
05/12/24-18:46:03 - Not triggering discharge as it is not requested
05/12/24-18:46:03 - Charging to and maintaining at 80% from 14%
05/12/24-18:46:03 - Charge below 80
05/12/24-18:46:03 - 🔌🔋 Enabling battery charging
05/12/24-18:46:03 - 🔼🪫 Disabling battery discharging
05/12/24-18:46:03 - 💡 Setting magsafe color to orange
05/12/24-19:31:05 - Charge above 80
05/12/24-19:31:05 - 🔌🪫 Disabling battery charging
05/12/24-19:31:05 - 💡 Setting magsafe color to green
05/15/24-15:00:33 - Not triggering discharge as it is not requested
05/15/24-15:00:33 - Charging to and maintaining at 80% from 16%
05/15/24-15:00:33 - Charge below 80
05/15/24-15:00:33 - 🔌🔋 Enabling battery charging
05/15/24-15:00:33 - 🔼🪫 Disabling battery discharging
05/15/24-15:00:33 - 💡 Setting magsafe color to orange
05/15/24-15:46:35 - Charge above 80
05/15/24-15:46:35 - 🔌🪫 Disabling battery charging
05/15/24-15:46:35 - 💡 Setting magsafe color to green
05/17/24-12:09:28 - Not triggering discharge as it is not requested
05/17/24-12:09:28 - Charging to and maintaining at 80% from 19%
05/17/24-12:09:28 - Charge below 80
05/17/24-12:09:28 - 🔌🔋 Enabling battery charging
05/17/24-12:09:28 - 🔼🪫 Disabling battery discharging
05/17/24-12:09:28 - 💡 Setting magsafe color to orange
05/17/24-12:54:30 - Charge above 80
05/17/24-12:54:30 - 🔌🪫 Disabling battery charging
05/17/24-12:54:30 - 💡 Setting magsafe color to green
05/17/24-14:39:43 - Charge below 80
05/17/24-14:39:43 - 🔌🔋 Enabling battery charging
05/17/24-14:39:43 - 🔼🪫 Disabling battery discharging
05/17/24-14:39:43 - 💡 Setting magsafe color to orange
05/22/24-12:28:08 - Not triggering discharge as it is not requested
05/22/24-12:28:08 - Charging to and maintaining at 80% from 20%
05/22/24-12:28:08 - Charge below 80
[... 136 lines cut ...]
Refreshing interface update timer
Display percentage 15 based on 16
Executing PATH=/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew/bin:/opt/homebrew/sbin:/opt/homebrew battery status_csv
Battery status: 
{"percentage":"16","remaining":"1:41","charging":true,"discharging":false,"maintain_percentage":"80","battery_state":"16% (1:41 remaining)","daemon_state":"smc charging enabled"}
Setting interface refresh speed to 0.5 minutes

📁 Config folder details:

total 392896
drwxr-xr-x@  6 jakjakob  staff   192B May 28 21:47 .
drwxr-x---+ 50 jakjakob  staff   1.6K May 28 21:52 ..
-rwxr-xr-x@  1 jakjakob  staff   227K May 28 21:52 battery.log
-rw-r--r--   1 jakjakob  staff     6B May 28 21:47 battery.pid
-rw-r--r--@  1 jakjakob  staff   177M May 28 21:52 gui.log
-rw-r--r--@  1 jakjakob  staff     3B May 28 21:47 maintain.percentage

⚙️  Battery data:

05/28/24-21:52:05 - Battery at 16% (1:41 remaining), smc charging enabled
05/28/24-21:52:05 - Your battery is currently being maintained at 80%
Battery CLI utility v1.1.6

</p>
</details>

jakjakob, 2024-06-02:
Update: the cause seems not to be valid, as rn at under 10% it shows up. Maybe only at 10% (it happened also)? But when charging, it shows it…

sebachuba, 2025-03-26:
[image]