actuallymentor/battery #282: enable smc charging not possible.

thiccasss, 2024-06-13:
After I disable battery limit it won't charge anymore and smc charging is still disabled.
If I write "battery charging on" in terminal, it still won't turn on and charge. Only way is to reset smc by restart the Mac.

how can I turn on smc charging again without restarting the Mac anytime?

[image]

mindok, 2024-06-15:
I had to uninstall to get charging past 80% going again. When using the CLI to try and turn on charging, I was getting the following output - I suspect the error of interest is `Error: SMCWriteKey() = e00002bc`:

`battery charging on`

```
06/15/24-16:38:22 - Setting charging to on
06/15/24-16:38:22 - Killing running maintain daemons & enabling charging as default state
06/15/24-16:38:22 - Disabling daemon at gui/501/com.battery.app
06/15/24-16:38:22 - 🔌🔋 Enabling battery charging
06/15/24-16:38:22 - 🔼🪫 Disabling battery discharging
06/15/24-16:38:22 - Charge above stop
06/15/24-16:38:22 - 🔌🪫 Disabling battery charging
06/15/24-16:38:22 - 💡 Setting magsafe color to green
Error: SMCWriteKey() = e00002bc
06/15/24-16:38:22 - 💡 Setting magsafe color to
Error: SMCWriteKey() = e00002bc
06/15/24-16:38:22 - Battery at 81% (attached; remaining), smc charging disabled
06/15/24-16:38:22 - 🔌🔋 Enabling battery charging
06/15/24-16:38:22 - 🔼🪫 Disabling battery discharging
06/15/24-16:38:22 - Charge above on
06/15/24-16:38:22 - 🔌🪫 Disabling battery charging
06/15/24-16:38:22 - 💡 Setting magsafe color to green
Error: SMCWriteKey() = e00002bc
```

I'm on a MacBook 13" M1 2020, macOS version 14.5. It broke on macOS version upgrade, but I can't recall which one as I haven't travelled for quite a few months.

adamlounds, 2024-06-25:
I had the same issue. Not sure why, but the `enable_charging` function _disables_ charging again ¯\_(ツ)_/¯

https://github.com/actuallymentor/battery/blob/481610ac05a6fedefb36d43032705cd9136f4691/battery.sh#L185
```shell
function enable_charging() {
	log "🔌🔋 Enabling battery charging"
	sudo smc -k CH0B -w 00
	sudo smc -k CH0C -w 00
	disable_discharging
}
```

To re-enable normal charging behaviour, try
```shell
battery maintain stop
sudo smc -k CH0B -w 00
sudo smc -k CH0C -w 00
```

lochstar, 2024-07-24:
I have the exact same issue.

papypaprika, 2024-10-26:
Same here.

```
battery maintain stop
10/26/24-11:38:32 - Killing old maintain process at 2779
10/26/24-11:38:32 - Killing running maintain daemons & enabling charging as default state
10/26/24-11:38:32 - Disabling daemon at gui/502/com.battery.app
10/26/24-11:38:32 - 🔌🔋 Enabling battery charging
10/26/24-11:38:32 - 🔼🪫 Disabling battery discharging
10/26/24-11:38:32 - Disabling discharging: No valid maintain percentage set, enabling charging
10/26/24-11:38:32 - MagSafe LED function invoked
10/26/24-11:38:32 - 💡 Setting magsafe color to orange
10/26/24-11:38:32 - setting LED to orange
Error: SMCWriteKey() = e00002bc
10/26/24-11:38:32 - Battery at 100% (0:00 remaining), 12.954V, smc charging enabled
```

swifty99, 2024-11-10:
same here M1Air 15.1. 
I am stuck in a calibration process. `battery charging on` is working though