actuallymentor/battery #287: 1.2.2, Always ask for backwards incompatible update after re-launch the app

sunnyxiong2020, 2024-06-29:
**What is the issue? (required)**
Installed the 1.2.2, quit app by "quit" button in the drop down menu, re-launch the app in the app folder.
Then the app asks for backwards update.every time

**What exactly did you do to produce the issue? (required)**
see above description

**Expected behavior (required)**
The backward update shall only happen once, not all the time

**Screenshots (optional)**
If applicable, add screenshots to help explain your problem. On a Mac you can click `control`+`command`+`shift`+`4` to create a screenshot that you can paste into this field with `command`+`4`.
[image]
[image]

**Error logs**
If you know how a terminal works, please paste the content of `battery logs` here. This allows me to see potential errors on your device.

Did `battery` fail to install? Use `tail -n 1000 ~/.battery/*/log` instead.

**Additional context (optional)**
Add any other context about the problem here.

dmatton, 2024-07-04:
Same problem for me.

Chip : M2 Pro
macOS : Sonoma 14.5

jparayre, 2024-07-05:
Same issue, battery.log contains a suspicious line : 

> 07/05/24-09:03:48 - Error: recover is not a valid setting for battery maintain. Please use a number between 0 and 100

Chip : M1 Max
macOS : Sonoma 14.5

NapoelonIV, 2024-07-08:
Same problem.
M3, macOS 14.5

blackketter, 2024-07-08:
Same problem here, m2 MacBook air.  I reset the log files and relaunched, new log files are attached.

[gui.log](https://github.com/user-attachments/files/16134622/gui.log)
[battery.log](https://github.com/user-attachments/files/16134623/battery.log)

blackketter, 2024-07-08:
I should also note that when I reboot, battery is not running and so my   hardware battery charges up.

NapoelonIV, 2024-07-09:
> I should also note that when I reboot, battery is not running and so my hardware battery charges up.

You have to add it to the startup items in settings in order for it to launch at boot.

Troubadixderdritte, 2024-07-16:
Same problem. I added battery to startup items... now it asks when I restart my computer. 
M3 Max, macOS 14.5

SvenRichter, 2024-07-25:
same issue... this is very annoying
MacBook Pro M2, MacOS 14.5

Andychengcoding, 2024-08-03:
Same issue.  It seems to keep trying to reach out to the setup script in this repo to keep reinstalling each time there's a login.  I looked at the setup script and it's innocuous enough but yeah, there's something wrong w/ the install if each it's trying to reinstall itself.

NapoelonIV, 2024-08-26:
Just to be clear, this bug has nothing to do with booting the computer: the bug occurs when the app loads. You can trigger it just by quoting and restarting the app. Also the behaviour of the bug alternates. First it asked for a password. and it says that it needs to install a backwards incompatible update. The next time the app is quit and reloaded it just displays the message without asking for the password.