actuallymentor/battery #289: Sailing mode to prevent trickle charging

jk-aus, 2024-07-01:
Battery does a great job of limiting charging or manually discharging to a set percentage but could a feature like Al Dente's sailing mode be added to prevent trickle charging when the laptop is plugged in and has reached the specified charging limit?  

Instead of just setting a "battery maintain" level, a charge limit level and a charge restart level would be set so the battery could drain down to a set point before charging restarts.
e.g. limit of 80% and restart point of 70% so the laptop charges to 80% and then uses power from the adapter but the battery is not being charged until its level slowly and naturally drops to 70%, and then charging restarts.

As it is, the laptop charges to 80% and then uses power from the adapter but is fairly constantly topping up the battery to keep it at 80%.

rguca, 2024-07-02:
As far as I understand you are looking for an adjustable hysteresis in percent. Currently it is coded at 1%.
I implemented it as second argument for voltages limits: 
E.g.
`battery maintain 11.4V 0.2V`
This will stop charging at 11.6V and restart at 11.2V.

It could also be implemented for percentages like so:
`battery maintain 75 5`
Which would lead to your example of 70%-80%

jk-aus, 2024-07-02:
Hi Rene, having looked up what adjustable hysteresis means, yes, that's what I'm talking about. It's good to know it's already there with a ±1% swing.

Are you saying v1.2.2 can already use the "battery maintain 11.4V 0.2V" instruction and that a future release could have the ability to use the "battery maintain 75 5" instruction?  If so, the percentage example seems less technical and more user-friendly than the voltage example. I think the percentage approach would be a valuable addition.

djbob2000, 2024-07-12:
> I implemented it as second argument for voltages limits

I miss displaying real voltage instead of percentages.