actuallymentor/battery #290: Battery calibration doesn't seem to work

Zagrthos, 2024-07-04:
**What is the issue? (required)**
The battery calibration CLI command doesn't fully charge the battery to 100%. Instead it stops at 83% and then disables the charging mechanism again.

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Restarted the MacBook
2. Started the Battery App
3. Opened the Terminal
4. Entered `battery calibrate`
5. Waited.

**Expected behavior (required)**
The battery is getting discharged to 15%, then charges up to 100%. Waits for an hour and then it discharges down until 80% again.

**Screenshots (optional)**
None.

**Error logs**
<details>
<summary>Open for logs</summary>
```
07/04/24-12:28:03 - 🚨 Calibration process have been stopped
07/04/24-12:28:03 - Error: recover is not a valid setting for battery maintain. Please use a number between 0 and 100
07/04/24-12:28:37 - 🚨 Calibration process have been stopped
07/04/24-12:28:37 - 🚨 Calibration process have been stopped
07/04/24-12:28:37 - Not triggering discharge as it is not requested
07/04/24-12:28:37 - Not triggering discharge as it is not requested
07/04/24-12:28:37 - Charging to and maintaining at 80% from 22%
07/04/24-12:28:37 - Charging to and maintaining at 80% from 22%
07/04/24-12:29:06 - Starting calibration
07/04/24-12:29:06 - 🚨 Calibration process have been stopped
07/04/24-12:29:06 - Killing running maintain daemons & enabling charging as default state
07/04/24-12:29:06 - Disabling daemon at gui/501/com.battery.app
07/04/24-12:29:06 - Battery at 22% ((no remaining), 11.25V, smc charging enabled
07/04/24-12:29:06 - Discharging to 15% from 22%
07/04/24-12:29:06 - 🔽🪫 Enabling battery discharging
07/04/24-12:29:06 - Battery at 22% (target 15%)
07/04/24-12:30:06 - Battery at 22% (target 15%)
07/04/24-12:31:06 - Battery at 22% (target 15%)
07/04/24-12:32:06 - Battery at 22% (target 15%)
07/04/24-12:33:06 - Battery at 22% (target 15%)
07/04/24-12:34:06 - Battery at 22% (target 15%)
07/04/24-12:35:06 - Battery at 21% (target 15%)
07/04/24-12:36:06 - Battery at 21% (target 15%)
07/04/24-12:37:06 - Battery at 21% (target 15%)
07/04/24-12:38:06 - Battery at 21% (target 15%)
07/04/24-12:39:06 - Battery at 21% (target 15%)
07/04/24-12:40:06 - Battery at 21% (target 15%)
07/04/24-12:41:06 - Battery at 21% (target 15%)
07/04/24-12:42:06 - Battery at 21% (target 15%)
07/04/24-12:43:06 - Battery at 21% (target 15%)
07/04/24-12:44:06 - Battery at 21% (target 15%)
07/04/24-12:45:06 - Battery at 21% (target 15%)
07/04/24-12:46:06 - Battery at 20% (target 15%)
07/04/24-12:47:53 - Battery at 20% (target 15%)
07/04/24-12:48:53 - Battery at 20% (target 15%)
07/04/24-12:49:54 - Battery at 20% (target 15%)
07/04/24-12:50:54 - Battery at 20% (target 15%)
07/04/24-12:51:54 - Battery at 20% (target 15%)
07/04/24-12:52:54 - Battery at 20% (target 15%)
07/04/24-12:53:54 - Battery at 20% (target 15%)
07/04/24-12:54:54 - Battery at 20% (target 15%)
07/04/24-12:55:54 - Battery at 20% (target 15%)
07/04/24-12:56:54 - Battery at 19% (target 15%)
07/04/24-12:57:54 - Battery at 19% (target 15%)
07/04/24-12:58:54 - Battery at 19% (target 15%)
07/04/24-12:59:54 - Battery at 19% (target 15%)
07/04/24-13:00:54 - Battery at 19% (target 15%)
07/04/24-13:01:54 - Battery at 19% (target 15%)
07/04/24-13:02:54 - Battery at 19% (target 15%)
[... 104 lines cut ...]
07/04/24-16:04:50 - Charge above 80
07/04/24-16:04:50 - MagSafe LED function invoked
07/04/24-16:04:50 - 💡 Setting magsafe color to green
07/04/24-16:04:50 - setting LED to green
07/04/24-16:05:50 - Charge above 80
07/04/24-16:05:50 - MagSafe LED function invoked
07/04/24-16:05:50 - 💡 Setting magsafe color to green
07/04/24-16:05:50 - setting LED to green
07/04/24-16:06:50 - Charge above 80
07/04/24-16:06:50 - MagSafe LED function invoked
07/04/24-16:06:50 - 💡 Setting magsafe color to green
07/04/24-16:06:50 - setting LED to green
07/04/24-16:08:13 - 🚨 Calibration process have been stopped
07/04/24-16:08:13 - 🚨 Calibration process have been stopped
07/04/24-16:08:13 - Not triggering discharge as it is not requested
07/04/24-16:08:13 - Not triggering discharge as it is not requested
07/04/24-16:08:13 - Charging to and maintaining at 80% from 83%
07/04/24-16:08:13 - Charging to and maintaining at 80% from 83%
07/04/24-16:08:16 - checking if at 100%
```

</details>

**Additional context (optional)**
None.

djbob2000, 2024-07-07:
For calibration you do not need to discharge, just charge to 100% within an hour.

Zagrthos, 2024-07-07:
Well I meant the calibration feature introduced with the new version.
This is supposed to discharge your battery down to 15%, then charging up to 100% and let it stay for an hour.

This is also inside the description of it.