actuallymentor/battery #291: battery.app fails to launch on M1 Air with Sonoma 14.5

jonathan-jbs, 2024-07-06:
The issue is that battery.app fails with error "Error installing battery limiter: undefined".

I have installed with brew, also by download of dmg, same result. 

I expected this to complete the installation and give me a CLI that runs.

[image]

**Error logs**
executable 'battery' does not exist.
File gui.log exists in ~/.battery... pasted below.

[ update-electron-app ] 
feedURL
https://update.electronjs.org/actuallymentor/battery/darwin-arm64/1.2.2
Starting tray app
Found image: /Applications/battery.app/Contents/Resources/battery-active-100-Template.png
Triggering boot-time auto-update
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin curl -I https://github.com &> /dev/null
[ update-electron-app ] 
checking-for-update
Display percentage 100 based on 100
[ update-electron-app ] 
requestHeaders
[object Object]
Tray app boot complete
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin curl -I https://icanhazip.com &> /dev/null
Get active logo for 100
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin which battery
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin sudo -n /usr/local/bin/smc -k CH0I -r
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin sudo -n /usr/local/bin/smc -k ACLC -w 02
Internet online: false
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin sudo -n /usr/local/bin/smc -k CH0C -r
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin sudo -n /usr/local/bin/smc -k ACLC -r
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin which smc
Is installed? 
false
Executing ps aux | grep "/usr/local/bin/battery " | wc -l | grep -Eo "\d*"
[ update-electron-app ] 
requestHeaders
[object Object]
[ update-electron-app ] 
feedURL
https://update.electronjs.org/actuallymentor/battery/darwin-arm64/1.2.2
Found image: /Applications/battery.app/Contents/Resources/battery-active-100-Template.png
Get active logo for 100
Display percentage 100 based on 100
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin curl -I https://github.com &> /dev/null
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin curl -I https://icanhazip.com &> /dev/null
[ update-electron-app ] 
checking-for-update
Tray app boot complete
Starting tray app
Triggering boot-time auto-update
Internet online: false
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin which battery
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin which smc
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin sudo -n /usr/local/bin/smc -k CH0C -r
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin sudo -n /usr/local/bin/smc -k CH0I -r
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin sudo -n /usr/local/bin/smc -k ACLC -r
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin sudo -n /usr/local/bin/smc -k ACLC -w 02
Is installed? 
false
Executing ps aux | grep "/usr/local/bin/battery " | wc -l | grep -Eo "\d*"
[ update-electron-app ] 
requestHeaders
[object Object]
[ update-electron-app ] 
feedURL
https://update.electronjs.org/actuallymentor/battery/darwin-arm64/1.2.2
[... 80 lines cut ...]
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin curl -I https://github.com &> /dev/null
[ update-electron-app ] 
checking-for-update
[ update-electron-app ] 
feedURL
https://update.electronjs.org/actuallymentor/battery/darwin-arm64/1.2.2
Get active logo for 100
Starting tray app
Internet online: false
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin which battery
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin sudo -n /usr/local/bin/smc -k ACLC -r
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin sudo -n /usr/local/bin/smc -k ACLC -w 02
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin sudo -n /usr/local/bin/smc -k CH0C -r
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin which smc
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin sudo -n /usr/local/bin/smc -k CH0I -r
Is installed? 
false
Executing ps aux | grep "/usr/local/bin/battery " | wc -l | grep -Eo "\d*"
Update/install error: 
/Users/jbs/.bashrc: line 5: epithet: command not found

[ update-electron-app ] 
update-not-available
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery status_csv
App initialisation process complete

jonathan-jbs, 2024-07-06:
The internet is working fine, this report filed on the same machine.

sglavoie, 2024-09-13:
I got roughly the same log outputs on M1 Sonoma 14.6.1 (also says `Internet online: false`), by installing with Homebrew and also with the standalone DMG file. The CLI version installed with the [installation script](https://github.com/actuallymentor/battery#installation-1) worked just fine for me, at least that's an option. Thank you!

pencilcheck, 2024-11-09:
I'm also getting this error on M1 air on Sonoma as well. Would be nice if the DMG works out of box.