actuallymentor/battery #294: Current limit depends on macOS charging percentage, which is not accurate.

js4jiang5, 2024-07-23:
**What is the issue? (required)**
Current limit depends on macOS charging percentage, which differs from real hardware charging percentage by 1~7%.

**What exactly did you do to produce the issue? (required)**
I recorded the charging percentages of macOS and CoconutBattery.
I set "battery maintain 65".
When macOS dropped to 64% and triggered charging, CoconutBattery shows 60.1%. The difference is about 4%.
After battery is charged to macOS 66%, CoconutBattery shows 65.1%. The difference is about 1%.
The difference varies, depending on whether it's charging or discharging.

**Expected behavior (required)**
As Al dente pointed out that they use real hardware charging percentage instead of macOS percentage, it would be better for battery.app to do the same for better accuracy.

**Additional context (optional)**
I tried to use voltage to remedy this issue, but found that even using voltage has problems.
For example, if you set "battery maintain 11.4V 0.2V", you expect it to trigger charging at 11.2V and stop charging at 11.6V.
However, in reality, when the charging is triggered at 11.2V, it abruptly jumped to 12.2V due to charging.

js4jiang5, 2024-09-01:
Since no one responded, I spent some time to solve it by myself by modifying battery.sh. The modifications include
1. replace macOS battery percentage with real hardware charging percentage.  
2. add sail mode with command "battery maintain 60 50", which stop charging when exceeding 60% and start charging when less than 50%
3. modify "battery callibrate" to discharge to maintain percentage instead of 80%
It's running well. Now the modified version is exactly what I needed. Bingo!

mtenjeri, 2024-09-01:
> Since no one responded, I spent some time to solve it by myself by modifying battery.sh. The modifications include
>
> 1. replace macOS battery percentage with real hardware charging percentage.
> 2. add [...]

Can you please share your modified version?

peteygao, 2024-09-02:
@js4jiang5 Care to create a PR with your changes?

js4jiang5, 2024-09-02:
> @js4jiang5 Care to create a PR with your changes?

I don't know how to PR, but I just checked current PR and found someone has been working on similar issue.

mtenjeri, 2024-09-02:
> > @js4jiang5 Care to create a PR with your changes?
> 
> I don't know how to PR, but I just checked current PR and found someone has been working on similar issue.

Tried that, doesn’t work as expected. 
Now we depend on you

js4jiang5, 2024-09-03:
> > > @js4jiang5 Care to create a PR with your changes?
> > 
> > 
> > I don't know how to PR, but I just checked current PR and found someone has been working on similar issue.
> 
> Tried that, doesn’t work as expected. Now we depend on you

Could you describe what went wrong?

mtenjeri, 2024-09-03:
> > > > @js4jiang5 Care to create a PR with your changes?
> > > 
> > > 
> > > I don't know how to PR, but I just checked current PR and found someone has been working on similar issue.
> > 
> > 
> > Tried that, doesn’t work as expected. Now we depend on you
> 
> Could you describe what went wrong?

Correction - it didn't work at all, if we talk about the same PR from user seefood. I don't remember the details, if you really need that info I could try again sometime this week.

js4jiang5, 2024-09-03:
> Correction - it didn't work at all, if we talk about the same PR from user seefood. I don't remember the details, if you really need that info I could try again sometime this week.

Yep, the PR from seefood. Please try again and let us know.

mtenjeri, 2024-09-14:
@js4jiang5 
Finally found some time, this is the output:

% battery maintain 70-80        
touch: /Users/USER/.battery/battery.log: Permission denied
/usr/local/bin/battery: line 935: syntax error: unexpected end of file

mihten, 2024-09-21:
I actually managed to use the solution from seefood, but adding his code manually to current battery.sh version. I lost some functionalities on the way (like maintain voltage), just for the sake of easier debugging, I don't even need it.
All in all, it's been working for the last week, maintaining percentage between 70 and 80%.

Maybe I'll try to merge them in a clean way, but I would like to see your solution too @js4jiang5

js4jiang5, 2024-10-19:
Don't use their calibrate. It is detrimental to the battery.
My battery health in coconutBattery app showed 100.2% when I purchased it in July this year.
[image]

After first use of calibrate, it dropped to 99.2%. I was kind of disappointed.
After 2nd use of calilbrate, it further dropped to 95.9%. I was totally shocked.
[image]
[image]

The reason I tried this battery app is because AlDente has two extreme comments on Reddit. One group of people praised it while the other group complained that their battery health dropped immedately at first use of AlDente.

I can't believe it also happened to me even with this battery app.
So I decided to fork it and develop my own version. Rewrite the calibrate function with integration of power control directly using smart home outlet switch.
Thank God. The battery health was back to 97.8% after first use of my own version, and even further improved back to 100.6% after 2nd use.
[image]
[image]

Still need to do more experiments to clarify something, though. I am always wondering what happened to AlDente that results in two extreme comments.

js4jiang5, 2024-12-31:
> I actually managed to use the solution from seefood, but adding his code manually to current battery.sh version. I lost some functionalities on the way (like maintain voltage), just for the sake of [...]

[my solution](https://github.com/js4jiang5/BatteryOptimizer_for_MAC)

js4jiang5, 2025-08-24:
> > I can't believe it also happened to me even with this battery app. So I decided to fork it and develop my own version. Rewrite the calibrate function with integration of power control directly using smart home outlet switch. Thank God. The battery health was back to 97.8% after first use of my own version, and even further improved back to 100.6% after 2nd use. [image] ![image](https://private-user-images.githubusercontent.com/89449908/378042759-188cba6e-ea61-40aa-a1cd-121f9e920855.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTYwMDA4MjAsIm5iZiI6MTc1NjAwMDUyMCwicGF0aCI6Ii84OTQ0OTkwOC8zNzgwNDI3NTktMTg4Y2JhNmUtZWE2MS00MGFhLWExY2QtMTIxZjllOTIwODU1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODI0VDAxNTUyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY5MTQwYTE2NGFiMzc2NzY1M2EwMjA1Y2NiMGI4ODE4ZWI2MzU5Y2MzYTlmMDNmYTc0OGJlZDYyNjAyMmFhYzImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.2FvnHicfS0sS9N938I43DWQX-gVNSPh5hMw7cUK9fXo)
> 
> [@js4jiang5](https://github.com/js4jiang5) How did you able to restore the health using your version of the app? What steps have you followed? I also face the same issue where my battery health decreased very quickly after using battery app

You may take a look at [my solution](https://github.com/js4jiang5/BatteryOptimizer_for_Mac). I've added many features and support Mac with Intel CPU. Just run "battery calibrate" after installation to check if it works or not.