actuallymentor/battery #298: Error installing battery limiter: Command failed: PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery maintain recover /bin/bash: battery: command not found

Barry1915, 2024-08-22:
Error installing battery limiter: Command failed: PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery maintain recover
/bin/bash: battery: command not found

Ox1deL, 2024-09-21:
same issue：
Error installing battery limiter: Command failed: PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery maintain recover
/bin/bash: battery: command not found

vapaa, 2024-10-01:
I have the same error with Sonoma 14.7

[image]
[image]

ColoCoffee, 2025-01-10:
I have the same error with Sonoma 15.2