actuallymentor/battery #300: Repeated query for Mac's password?

tvirku, 2024-08-30:
I just updated to 1.2.2. Thanks for a useful app! But, since then, I'm continually asked for my Mac's password if I've booted the Mac. This is different to the past.  Is this OK?

lcsvcn, 2024-09-03:
same issue, after the update it is asking several times for password and also not starting with macos

diazmed, 2024-09-05:
Same issue.. almost every time it says "Battery needs to apply a backwards incompatible update, to do this it will ask for your password. This should not happen frequently."

but sometimes it starts with another error saying Internet not found and needed for updates

sqrt9, 2024-09-11:
here from google. also having this issue

diazmed, 2024-09-11:
@sqrt9 Just uninstall 1.22 and install 1.21.. worked fine for me

tollfree, 2024-09-22:
Run `battery visudo` once

diazmed, 2024-09-23:
@tollfree it just says "The existing battery visudo file is what it should be for version v1.2.7"

fergymcferguson, 2024-09-27:
bump - same here for every reboot and every time the program is stopped and restarted.

txrunn, 2024-10-02:
bump - same here, unsure if related to recent sonoma upgrade

rodalpho, 2024-10-07:
This is extremely annoying, as touchID doesn't work to authenticate.

Ca-Wi, 2024-10-08:
Here also ... Beta from today