actuallymentor/battery #301: Battery calibrate does not work!

tvirku, 2024-09-06:
I liked to calibrate my MBP '23's battery using the battery calibrate command. It didn't work.

So, I used the command in the terminal (shown by the battery command). The electric cord was attached, and its light remained dimmed, as expected. As I was working with MBP, I noticed that the charge dropped to 15%, but stayed there for an unexpected long time. Charging was not started. I disconnected and connected the cord, and charging began, according to the orange light in the connector. However, for some 30 min, the percentage of charge did not rise.

In this phase, I quitted the battery app, disconnected/connected the cord, and charging began very soon. Now it is 85%.

In conclusion, I was not able to use Battery for calibrating my MBP's battery. I hope that the MBP's battery is calibrated by just charging the device normally up to 100%.  In all, I like the app for its basic use, although after the latest update, it is frustrating that I need to insert my password after every booting.

js4jiang5, 2024-10-18:
Don't use their calibrate function because there are severe bugs and it's detrimental to your battery. Apparently it was released without testing first. Just use their maintain function.

natecodes, 2024-11-18:
+1 exact same issue, drains down to 15% doesn't charge up