actuallymentor/battery #307: battery adapter is not working in Sonoma

gk4code1, 2024-09-27:
**What is the issue? (required)**
Battery adapter off is not turning off the charging.

I see in the logs error message as Disabling discharging: No valid maintain percentage set, enabling charging

I tried setting battery maintain 80  but still same error

Battery app version is 1.2.2

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Go to terminal type battery adapter off 
2. See error

**Expected behavior (required)**
it should turn off the charging.

**Error logs**
➜  ~ battery adapter off                                                                           ⚡️
09/27/24-10:17:34 - Setting adapter to off
09/27/24-10:17:34 - Killing running maintain daemons & enabling charging as default state
09/27/24-10:17:34 - Disabling daemon at gui/501/com.battery.app
09/27/24-10:17:34 - 🔌🔋 Enabling battery charging
09/27/24-10:17:34 - 🔼🪫 Disabling battery discharging
09/27/24-10:17:34 - Disabling discharging: No valid maintain percentage set, enabling charging
09/27/24-10:17:34 - MagSafe LED function invoked
09/27/24-10:17:34 - 💡 Setting magsafe color to orange
09/27/24-10:17:34 - setting LED to orange
09/27/24-10:17:34 - Battery at 100% (0:00 remaining), 12.892V, smc charging enabled
09/27/24-10:17:34 - 🔼🪫 Disabling battery discharging
09/27/24-10:17:34 - Disabling discharging: No valid maintain percentage set, enabling charging
09/27/24-10:17:34 - MagSafe LED function invoked
09/27/24-10:17:34 - 💡 Setting magsafe color to orange
09/27/24-10:17:34 - setting LED to orange

tail -n 1000 ~/.battery/battery.log

09/24/24-20:37:52 - No setting to recover, exiting
09/24/24-20:37:53 - Not triggering discharge as it is not requested
09/24/24-20:37:53 - Charging to and maintaining at 80% from 100%
09/24/24-20:37:53 - Charge above 80
09/24/24-20:37:53 - 🔌🪫 Disabling battery charging
09/24/24-20:37:53 - MagSafe LED function invoked
09/24/24-20:37:53 - 💡 Setting magsafe color to green
09/24/24-20:37:53 - setting LED to green
09/24/24-20:38:02 - Not triggering discharge as it is not requested
09/24/24-20:38:02 - Charging to and maintaining at 80% from 100%
09/24/24-20:38:40 - Not triggering discharge as it is not requested
09/24/24-20:38:40 - Charging to and maintaining at 80% from 100%

rkrug, 2024-10-01:
It seems to be limiting the charging to 80% and setts the led to green, but the menubar icon is not shown.

vitiasr, 2024-10-06:
It works but in reverse. In order to disable charging you need to type `battery adapter on`. Typing `battery adapter off` will resume charging. This bug is present in version 1.2.7. Actually, after updating to v1.2.7, `battery charge` and `battery maintain` stopped working.

rkrug, 2024-10-07:
Just checked again - now the menue bar icon is shown. Must have been a hiccup in my system.

avdok, 2024-10-09:
I had the same problem. The icon in the menu bar was not displayed. I installed via brew and dmg file. After rebooting, I launched the battery app manually. Closed it via ctrl-q and launched it again. The application again prompted to enter the password and the icon in the menu bar appeared.

PS: If you use the Hidden Bar app, it's worth expanding the hidden icons part.