actuallymentor/battery #312: Battery requests to install a backwards incompatible update every time the app is started on MacOS Sequoia

wild-a, 2024-10-25:
**What is the issue? (required)**
Battery requests to install a backwards incompatible update every time the app is started on MacOS Sequoia.

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Ensure you are on MacOS 15.0 or greater
2. Start Battery (quit and restart if already running).
3. A message appears saying "Battery needs to apply a backwards incompatible update, to do this it will ask for your password. This should not happen frequently."
4. Something oss requests a change to system with password.
5. After everything is done, Battery runs normally.
6. However, if you quit it, it goes back to step 3.

**Expected behavior (required)**
I expect battery to install that required item once and not again.

**Screenshots (optional)**
[image]

conjon42, 2024-10-26:
I have the same issue. I installed with Homebrew cask on latest version of MacOS Sonoma

CHTJonas, 2024-10-26:
Ditto here with v1.2.2 installed via Homebrew on Sonoma 14.7 (23H124).

CHTJonas, 2024-10-26:
Dupe of #300.

deyshin, 2024-11-06:
Same issue on my side with Sequoia 15.1

tyhallcsu, 2024-11-07:
Same. Everytime I restart the battery.app or reboot.

First it asks for permission, then gives a password prompt:
[image]

Then after it does it's thing, this is the confirmation message.
[image]

MacBook Pro | Apple M2 Max | Sonoma 14.6.1

tyhallcsu, 2024-11-07:
# Battery App Password-Free Setup Script

If you're tired of entering your password every time the Battery app starts up, this script will configure your system to allow the Battery app to run without password prompts.

## What This Script Does

- Creates a backup of your sudoers configuration
- Adds specific permissions for the Battery app commands
- Tests the configuration to ensure it works
- Provides detailed logging of all actions
- Includes safety checks and error handling

## How to Use

1. Create a new file called `fix-battery.sh`
2. Copy the script below into the file
3. Make it executable: `chmod +x fix-battery.sh`
4. Run it with sudo: `sudo ./fix-battery.sh`

## The Script

```bash
#!/bin/bash

# Color codes for output
GREEN='\033[0;32m'
RED='\033[0;31m'
YELLOW='\033[1;33m'
NC='\033[0m' # No Color

# Get current user
CURRENT_USER=$(whoami)
if [ "$CURRENT_USER" = "root" ]; then
    CURRENT_USER=$SUDO_USER
fi

# Logging function
log() {
    echo -e "${YELLOW}[$(date '+%Y-%m-%d %H:%M:%S')]${NC} $1"
}

error() {
    echo -e "${RED}[$(date '+%Y-%m-%d %H:%M:%S')] ERROR:${NC} $1"
}

success() {
    echo -e "${GREEN}[$(date '+%Y-%m-%d %H:%M:%S')] SUCCESS:${NC} $1"
}

# Check if script is run as root
if [ "$EUID" -ne 0 ]; then 
    error "Please run this script with sudo"
    exit 1
fi

# Print script start information
log "Starting battery sudo configuration script"
log "Configuring for user: $CURRENT_USER"

# Create backup of sudoers
log "Creating backup of sudoers configuration..."
if cp /etc/sudoers /etc/sudoers.backup; then
    success "Backup created at /etc/sudoers.backup"
else
    error "Failed to create sudoers backup"
    exit 1
fi

# Create new sudoers file for battery app
log "Creating battery app sudoers configuration..."
cat > /etc/sudoers.d/battery-app << EOL
$CURRENT_USER ALL=(ALL) NOPASSWD: /usr/local/bin/smc -k CH0C -r
$CURRENT_USER ALL=(ALL) NOPASSWD: /usr/local/bin/smc -k CH0I -r
$CURRENT_USER ALL=(ALL) NOPASSWD: /usr/local/bin/smc -k ACLC -r
$CURRENT_USER ALL=(ALL) NOPASSWD: /usr/local/bin/smc -k ACLC -w 02
$CURRENT_USER ALL=(ALL) NOPASSWD: /usr/local/bin/battery
EOL

# Set correct permissions
log "Setting correct permissions..."
if chmod 440 /etc/sudoers.d/battery-app; then
    success "Permissions set correctly"
else
    error "Failed to set permissions"
    rm /etc/sudoers.d/battery-app
    exit 1
fi

# Validate sudoers syntax
log "Validating sudoers syntax..."
if visudo -c -f /etc/sudoers.d/battery-app; then
    success "Sudoers syntax validated successfully"
else
    error "Sudoers syntax validation failed"
    rm /etc/sudoers.d/battery-app
    exit 1
fi

# Test the configuration
log "Testing configuration with actual commands..."

# Test array of commands
declare -a test_commands=(
    "/usr/local/bin/smc -k CH0C -r"
    "/usr/local/bin/smc -k CH0I -r"
    "/usr/local/bin/smc -k ACLC -r"
)

# Run tests
TEST_FAILED=0
for cmd in "${test_commands[@]}"; do
    log "Testing command: $cmd"
    if sudo -n $cmd &>/dev/null; then
        success "Command executed without password: $cmd"
    else
        error "Failed to execute without password: $cmd"
        TEST_FAILED=1
    fi
done

if [ $TEST_FAILED -eq 0 ]; then
    echo
    echo -e "${GREEN}✅ Battery sudo configuration completed successfully${NC}"
    echo -e "${GREEN}✅ All test commands executed without password prompts${NC}"
    echo
    log "You can now use the battery app without password prompts"
    log "If you need to undo these changes, run: sudo rm /etc/sudoers.d/battery-app"
else
    echo
    error "Some tests failed. Configuration may not be complete."
    error "Check the logs above for details"
    echo
    exit 1
fi

# Print final status check
echo
log "Final Configuration Status:"
echo "-------------------------"
echo "User configured: $CURRENT_USER"
echo "Sudoers file: /etc/sudoers.d/battery-app"
echo "Backup file: /etc/sudoers.backup"
echo "Syntax validation: Passed"
echo "Command tests: Passed"
echo "-------------------------"
```

## How to Undo Changes

If you need to remove these permissions, simply run:
```bash
sudo rm /etc/sudoers.d/battery-app
```

## Safety Note

This script modifies system permissions to allow the Battery app to run specific commands without a password. While this is safe for the Battery app's specific commands, you should always be careful when modifying sudo permissions on your system.

The script includes several safety features:
- Creates a backup of your original configuration
- Validates syntax before applying changes
- Tests the configuration before completing
- Can be easily reverted if needed

Morgan1er, 2025-01-02:
Hi, same problem here : MacBook Pro 16 M1 Max with Sequioa 15.2.
I didn't try the script posted before, I'd prefer a proper integration with the app. 
Thx for your help.

Griefed, 2025-01-08:
Looking forward to the fix. I am experiencing the same issue with an M4 and Sequoia 15.2. Thanks for the hard work!

Sevilum, 2025-01-13:
Same issue for me, sadly.
System:
Macbook Pro 16 M1 Pro 
Sequoia 15.2
Battery v1.2.2

teqtic, 2025-01-21:
Also seeing this with every reboot after adding this app to my login items so that it starts on startup. MacOS Sequoia 15.2 on M1 Max.

tyhallcsu, 2025-01-21:
Issue hasn’t been fixed, switched to Al Dente; will return once the issue is fixed

saminitz, 2025-02-04:
I'm also having the same issue on v1.2.2 with Sequoia 15.3 on M4 Max

haraldt1, 2025-02-08:
same here, fix doesn't work MacBookmPro M3pro Sequoia 15.2

fametrano, 2025-04-30:
any chance to fix this annoying bug?

tyhallcsu, 2025-05-01:
I have switched to using AlDente; would love to switch back after the issue has been fixed.

PS -- the script I wrote was experimental and didnt work. I see so much traffic here im concerned many of you actually used it.

noclascio, 2025-05-12:
+1  Back to aldente