actuallymentor/battery #321: After deinstallation, battery wont charge to 100.

ghostforest, 2024-12-27:
**What is the issue? (required)**
I deinstalled battery and it still stops at 80%. I set 80 to be the maximum with battery. However now its gone and it still stops at 80.
Id like to fix this. 
I also recognized that "battery charge 80" would never fully execute in terminal. It seemed to get stuck. Thats why I uninstalled it. I downloaded Aldente and set the battery to 100 and now it charges to 100. But as soon al dente is gone my battery will again stop at 80%. 

What is this.

il2020, 2025-01-08:
I have the same problem. I wanted to charge the battery to 100% today, but I couldn't do it.

Vali-Mandeal, 2025-01-20:
@ghostforest you could try the following command:

sudo pmset restoredefaults

This should restore the battery configurations to default.

il2020, 2025-01-20:
> sudo pmset restoredefaults

It didn't help me.

ghostforest, 2025-01-20:
Deinstall with Battery and then install aldente. install and deinstall too. that fixed it for me.

Vali-Mandeal, 2025-01-20:
> > sudo pmset restoredefaults
> 
> It didn't help me.

So on a M1 Pro with the latest OS Version I did the following (in this particular order):
1) Disable battery charge limit with this software
2) Uninstall the software 
3) Run the command

You could also try the Aldente suggestion from this thread.

il2020, 2025-01-21:
I got it, thank you. But I should have made sure to click on “Charge to Full Now” here.

[image]

Gazmedbretann, 2025-03-18:
Hello, 
PLease forgive my low and medium mastery of internet tech. and English languages, respectively
I had this problem after I uninstalled Battery
I could prevent it from running in the settings. 
Yet I get warnings on its presence 
MacBookAir , Sequoia. 
Should I reinstall it set it to "charge to full" then un install it ? 

[Why did I uninstall it ? Becquse I thought it could give warnings at 20% and 80% charge values. ]

Thanks for your help

Gazmed