actuallymentor/battery #327: Error installing battery limiter: Command failed: PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery maintain recover /bin/bash: battery: command not found

bobweein, 2025-01-16:
when i install this app ,it shows "Error installing battery limiter: Command failed: PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery maintain recover
/bin/bash: battery: command not found
"

jamestangeres, 2025-01-23:
same issue in mac m1

Battery limiter error: Command failed: PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery status_csv
/Users/james/.bashrc: line 2: pyenv: command not found
/Users/james/.bashrc: line 3: pyenv: command not found
/bin/bash: battery: command not found

Desmond4u, 2025-02-18:
the same in mac m1pro
Error installing battery limiter: Command failed: PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery maintain recover
/bin/bash: battery: command not found

MinCheng7, 2025-03-12:
Same error in MacBook Air M4
"Error installing battery limiter: Command failed: PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery maintain recover
/bin/bash: battery: command not found"

Xiaopang-yhq, 2025-03-17:
Error installing battery limiter: Command failed: PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery maintain recover
/bin/bash: battery: command not found
Same error in M1 Pro

ratio-dd, 2025-08-15:
same error in M4 Pro