actuallymentor/battery #329: Battery charging beyond 80% despite limiter being active

DenisTheProgrammer, 2025-01-22:
**What is the issue? (required)**
On my first charge using the app, the battery charged to 84% while I had the 80% limit active

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Log out of account
2. Plug in charger cable
3. Close the lid

**Expected behavior (required)**
Battery should have stopped charging at 80%

EmanueleSpadaro, 2025-01-24:
Same, it always fluctuates from 78% to 82%.

[image]

DenisTheProgrammer, 2025-01-26:
from what I discovered after more use, whenever the lid is closed the app seems to not register the actual % of the battery as it charges and gets stuck on whatever the battery was last on until you click it, however when you're using the mac and charging at the same time it seems to work