actuallymentor/battery #332: It reaches 100% when it is closed.

emresaracoglu, 2025-02-03:
Hi,

Thank you for this project. When the computer is asleep/off, the charge reaches 100% again. How can I keep the charge level at 80% even if the machine is turned off?

teqtic, 2025-02-13:
Unfortunately this was my biggest issue with this app as well. If the computer was already sleeping and you plug it in, it keeps on charging. I haven't had this issue with this fork:

https://github.com/js4jiang5/BatteryOptimizer_for_MAC

mrpacogp, 2025-02-16:
Maybe if you use an external monitor also rebase 80%. of charging?

emresaracoglu, 2025-02-16:
@teqtic Thanks for the suggestion! I will give it a try.

@mrpacogp I am using an external monitor, but the charge still reaches 100%. Also, how could the use of an external monitor be related to this?

mrpacogp, 2025-02-16:
> [@mrpacogp](https://github.com/mrpacogp) I am using an external monitor, but the charge still reaches 100%. Also, how could the use of an external monitor be related to this?

im wrong. im using an external monitor with thunderbolt, and the macbook get power from this, so i can unplug the power connector and the macbook still at 80% every time. I think the problem ebecome when i set the mbp to sleep then is when rebases the 80% (this morning was at 82%) maybe its a minor detail. i apologize for my english. right now is working fine, but mine is open is not closed.

emresaracoglu, 2025-02-16:
@mrpacogp 	Now I understand, have you observed how the charge level behaves when the MacBook is completely powered off?

mrpacogp, 2025-02-17:
> [@mrpacogp](https://github.com/mrpacogp) Now I understand, have you observed how the charge level behaves when the MacBook is completely powered off?

I havent powered off jet my laptop but i will do it tihs night, regards.

emresaracoglu, 2025-02-17:
I hope it works, regards. @mrpacogp