actuallymentor/battery #334: Always opens with the same error

rootstudio247, 2025-02-23:
Battery 1.2.2 never runs when I start my MBP M4. When I manually start it up, I always get the message that it needs to install some backward compatibility stuff and will ask for my password, but this shouldn't happen very often. Well, it happens every time. And my Mac is always at 100% charge. BTW, I run it in clamshell mode most of the time, connected to an external monitor with a magic keyboard and a BT mouse.

rootstudio247, 2025-02-25:
Btw, today I woke my Mac from sleep mode and got a popup message from Battery: Unknown error. Not sure if this software really works.

quadZ-z-Z, 2025-03-13:
Pump: I'm also experiencing this, v1.2.2 with MBP M1 Pro. Is there a way to permanently give it permission to do its thing so I don't have to re-input it every time I reboot?/

js4jiang5, 2025-03-13:
The latest version is v1.2.7. Why do you guys install v1.2.2?

melyux, 2025-03-17:
@js4jiang5 Where are you seeing 1.2.7?

js4jiang5, 2025-03-17:
> [@js4jiang5](https://github.com/js4jiang5) Where are you seeing 1.2.7?

[image]

rootstudio247, 2025-03-18:
Maybe it's platform-specific? When I click on 'Check for updates' in the UI, it takes me to a page with 1.2.2 as the latest version, more specifically this page:
https://github.com/actuallymentor/battery/releases

js4jiang5, 2025-03-18:
> Maybe it's platform-specific? When I click on 'Check for updates' in the UI, it takes me to a page with 1.2.2 as the latest version, more specifically this page: https://github.com/actuallymentor/battery/releases

I have developed my version based on v1.2.7 trying to solve all issues. You may give it shot if you want. My version is [here](https://github.com/js4jiang5/BatteryOptimizer_for_MAC).  Remember to setup the settings after installation to receive notifications successfully.

victorlin, 2025-04-24:
Duplicate of #312