actuallymentor/battery #351: Charging limit not working after update

DandeMC, 2025-07-01:
Hello,

since the last update, the charge limit no longer works for me. my macbook now charges to 100% and the unload function doesn't work either.

👾 Battery CLI logs:

07/01/25-15:32:29 - Debug trail. User: USER, config folder: /Users/USER/.battery, logfile: /Users/USER/.battery/battery.log, file called with 1: maintain_synchronous, 2: recover
07/01/25-15:32:29 - No setting to recover, exiting
07/01/25-15:32:30 - Triggering discharge to 80 before enabling charging limiter
07/01/25-15:32:30 - Discharging to 80% from 100%
07/01/25-15:32:30 - 🔽🪫 Enabling battery discharging
Error: SMCWriteKey() = e00002bc
07/01/25-15:32:30 - Battery at 100% (target 80%)
07/01/25-15:33:30 - Battery at 100% (target 80%)
07/01/25-15:34:30 - Battery at 100% (target 80%)
07/01/25-15:35:30 - Battery at 100% (target 80%)
07/01/25-15:36:30 - Battery at 100% (target 80%)
07/01/25-15:36:31 - Not triggering discharge as it is not requested
07/01/25-15:36:31 - Charging to and maintaining at 80% from 100%
07/01/25-15:37:06 - Triggering discharge to 80 before enabling charging limiter
07/01/25-15:37:06 - Discharging to 80% from 100%
07/01/25-15:37:06 - 🔽🪫 Enabling battery discharging
Error: SMCWriteKey() = e00002bc
07/01/25-15:37:06 - Battery at 100% (target 80%)
07/01/25-15:37:30 - Battery at 100% (target 80%)
07/01/25-15:38:06 - Battery at 100% (target 80%)
07/01/25-15:38:30 - Battery at 100% (target 80%)
07/01/25-15:39:06 - Battery at 100% (target 80%)

🖥️	Battery GUI logs:

Interface enabled limiter, percentage remaining: 100
Refreshing tray icon...
Battery status:
{"percentage":"100","remaining":"0:00","charging":false,"discharging":true,"maintain_percentage":"80","battery_state":"100% (0:00 remaining)","daemon_state":"forcing discharge to 80%"}
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery status
Limiter status message:
07/01/25-15:37:08 - Battery at 100% (0:00 remaining), 13.07V, smc charging disabled
07/01/25-15:37:08 - Your battery is currently being maintained at 80%

Force discharge setting: boolean true
Generate app menu percentage: 100 (discharge allowed, limited on)
Found image: /Applications/battery.app/Contents/Resources/battery-active-100-Template.png
Resetting interface timer speed
Display percentage 100 based on 100
Get active logo for 100
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery status_csv
Refreshing interface update timer
Battery status:
{"percentage":"100","remaining":"0:00","charging":false,"discharging":true,"maintain_percentage":"80","battery_state":"100% (0:00 remaining)","daemon_state":"forcing discharge to 80%"}
Setting interface refresh speed to 0.5 minutes
Refreshing tray icon...
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery status_csv
Battery status:
{"percentage":"100","remaining":"0:00","charging":false,"discharging":true,"maintain_percentage":"80","battery_state":"100% (0:00 remaining)","daemon_state":"forcing discharge to 80%"}
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery status
Force discharge setting: boolean true
Limiter status message:
07/01/25-15:37:38 - Battery at 100% (0:00 remaining), 13.071V, smc charging disabled
07/01/25-15:37:38 - Your battery is currently being maintained at 80%

Display percentage 100 based on 100
Found image: /Applications/battery.app/Contents/Resources/battery-active-100-Template.png
Get active logo for 100
Resetting interface timer speed
Refreshing interface update timer
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery status_csv
Generate app menu percentage: 100 (discharge allowed, limited on)
Setting interface refresh speed to 0.5 minutes
Battery status:
{"percentage":"100","remaining":"0:00","charging":false,"discharging":true,"maintain_percentage":"80","battery_state":"100% (0:00 remaining)","daemon_state":"forcing discharge to 80%"}
Refreshing tray icon...
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery status_csv
Battery status:
{"percentage":"100","remaining":"0:00","charging":false,"discharging":true,"maintain_percentage":"80","battery_state":"100% (0:00 remaining)","daemon_state":"forcing discharge to 80%"}
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery status
Limiter status message:
07/01/25-15:38:09 - Battery at 100% (0:00 remaining), 13.071V, smc charging disabled
07/01/25-15:38:09 - Your battery is currently being maintained at 80%

Generate app menu percentage: 100 (discharge allowed, limited on)
Display percentage 100 based on 100
Force discharge setting: boolean true
Get active logo for 100
Refreshing interface update timer
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery status_csv
Resetting interface timer speed
Found image: /Applications/battery.app/Contents/Resources/battery-active-100-Template.png
Battery status:
{"percentage":"100","remaining":"0:00","charging":false,"discharging":true,"maintain_percentage":"80","battery_state":"100% (0:00 remaining)","daemon_state":"forcing discharge to 80%"}
Setting interface refresh speed to 0.5 minutes
Refreshing tray icon...
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery status_csv
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery status
Battery status:
{"percentage":"100","remaining":"0:00","charging":false,"discharging":true,"maintain_percentage":"80","battery_state":"100% (0:00 remaining)","daemon_state":"forcing discharge to 80%"}
Limiter status message:
07/01/25-15:38:39 - Battery at 100% (0:00 remaining), 13.07V, smc charging disabled
07/01/25-15:38:39 - Your battery is currently being maintained at 80%

Get active logo for 100
Generate app menu percentage: 100 (discharge allowed, limited on)
Display percentage 100 based on 100
Found image: /Applications/battery.app/Contents/Resources/battery-active-100-Template.png
Refreshing interface update timer
Resetting interface timer speed
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery status_csv
Force discharge setting: boolean true
Battery status:
{"percentage":"100","remaining":"0:00","charging":false,"discharging":true,"maintain_percentage":"80","battery_state":"100% (0:00 remaining)","daemon_state":"forcing discharge to 80%"}
Setting interface refresh speed to 0.5 minutes
Refreshing tray icon...
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery status_csv
Battery status:
{"percentage":"100","remaining":"0:00","charging":false,"discharging":true,"maintain_percentage":"80","battery_state":"100% (0:00 remaining)","daemon_state":"forcing discharge to 80%"}
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery status
Force discharge setting: boolean true
Display percentage 100 based on 100
Generate app menu percentage: 100 (discharge allowed, limited on)
Found image: /Applications/battery.app/Contents/Resources/battery-active-100-Template.png
Resetting interface timer speed
Limiter status message:
07/01/25-15:39:09 - Battery at 100% (0:00 remaining), 13.07V, smc charging disabled
07/01/25-15:39:09 - Your battery is currently being maintained at 80%

Refreshing interface update timer
Executing PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin battery status_csv
Get active logo for 100
Battery status:
{"percentage":"100","remaining":"0:00","charging":false,"discharging":true,"maintain_percentage":"80","battery_state":"100% (0:00 remaining)","daemon_state":"forcing discharge to 80%"}
Setting interface refresh speed to 0.5 minutes

📁 Config folder details:

total 112
drwxr-xr-x@  7 USER  staff   224B  1 Jul 15:37 .
drwxr-x---+ 31 USER  staff   992B  1 Jul 15:35 ..
-rwxr-xr-x@  1 USER  staff   1,3K  1 Jul 15:39 battery.log
-rw-r--r--@  1 USER  staff     6B  1 Jul 15:37 battery.pid
-rw-r--r--@  1 USER  staff    37K  1 Jul 15:39 gui.log
-rw-r--r--@  1 USER  staff     3B  1 Jul 15:37 maintain.percentage
-rw-r--r--   1 root   staff   965B  1 Jul 15:32 visudo.tmp

⚙️	Battery data:

07/01/25-15:39:25 - Battery at 100% (0:00 remaining), 13.07V, smc charging disabled
07/01/25-15:39:25 - Your battery is currently being maintained at 80%
Battery CLI utility v1.2.7

karolpeszek, 2025-07-13:
Have you ever installed macOS 26 Beta? Even on separate volume on your disk or a USB stick?

iammicloud, 2025-07-13:
[This fork](https://github.com/js4jiang5/BatteryOptimizer_for_Mac) has solved the issue. Works well in macOS 26 Tahoe.

DandeMC, 2026-10-28:
Quick update for anyone landing here: I solved the problem by simply buying AlDente, which handles this just fine.

Since the developer apparently has no interest in responding to this issue, I'm closing it now. I'm no longer interested in pursuing this any further.

closed 2026-10-28 by DandeMC