actuallymentor/battery #355: Support for dual charge thresholds (e.g. start below 40%, stop at 80%)

Aquarius58, 2025-07-19:
I’d like to propose a feature that would greatly improve battery health management for macOS users who use external power sources like USB-C monitors:

> **Allow definition of two thresholds:**
> - **Upper threshold** (e.g. 80%): charging stops here (already supported by `maintain`)
> - **Lower threshold** (e.g. 40%): charging only starts again once battery drops below this

Currently, `battery maintain 80` is very useful to avoid overcharging. However, whenever I connect my MacBook to a USB-C monitor or any other power source, charging resumes immediately — even if the battery is at 79%. This behavior causes unnecessary micro-charging cycles, which can **reduce battery longevity**.

With dual-threshold logic (80/40), the battery would:
- Charge up to 80% (as now),
- But **not start charging again until it drops below 40%**.

Thanks and regards,
Thomas

spacesick, 2025-08-12:
Can you cite some research to prove your claim? Because I've read an experimental study that shows micro-charging can actually prolong battery life: [https://www.sciencedirect.com/science/article/pii/S2352152X2201338X](https://www.sciencedirect.com/science/article/pii/S2352152X2201338X)

Vossera, 2025-11-25:
Hello, I recomend you to use the cli command, just
```
battery charging off
```
This command would solve your problem.
The author said that GUI version would remain minmal set of function. The only way is to use command line. Or you can use my project of battery-plus, just a GUI-plus version of this project.