actuallymentor/battery #360: Feature Request: Add Lower Limit for Charging Resume

tigris030, 2025-08-08:
First of all, thank you for building this amazing tool!

Currently, the tool allows setting an upper limit for charging (e.g., stop at 50%), but there’s no way to define a lower limit at which charging should start again.

I would like to be able to set two limits:
- Upper limit → Stop charging at e.g. 50%
- Lower limit → Resume charging at e.g. 35%

This would help reduce unnecessary charging cycles and keep the battery in an optimal range for longevity. Right now, once the battery drains even slightly below the upper limit, it starts charging again, which causes more frequent top-up cycles.

Example:
- Set upper limit: 50%
- Set lower limit: 35%
- Battery charges to 50%, then stops.
- Once the charge drops to 35%, charging resumes until 50%.

Possible implementation ideas:
- Add a CLI flag or configuration option for the lower limit (similar to the existing upper limit setting).
- Ensure the monitoring loop checks both limits before deciding to enable or disable charging.

Would love to hear your thoughts on whether this could be added!

iammicloud, 2025-08-08:
Already available in [this fork](https://github.com/js4jiang5/BatteryOptimizer_for_Mac).

tigris030, 2025-08-08:
> Already available in [this fork](https://github.com/js4jiang5/BatteryOptimizer_for_Mac).

I don't like some things with this fork.