actuallymentor/battery #368: Does not work with MacOS 26 (Tahoe)

ronit18, 2025-08-20:
On macos 26 Public beta, this is not working.

cha0o0, 2025-08-29:
It not working in my macOS 26.0 beta device too

TeeAaTeeUu, 2025-09-16:
At issue https://github.com/actuallymentor/battery/issues/348 people also did some good research on the new updated SMC keys and values. Similarly https://github.com/js4jiang5/BatteryOptimizer_for_Mac added these two changes to support new version: https://github.com/js4jiang5/BatteryOptimizer_for_Mac/commit/42e3eb5ba6820e947f109c1e6dd3dfd388d8ecd1 and https://github.com/js4jiang5/BatteryOptimizer_for_Mac/commit/c0bb6869cddef228ae04b3a82349442b2ffba5fc .

Seemingly the only added command is `smc -k CHTE -w 01000000` to disable charging, along with the existing SMC key changes. Uncertain how AlDente achieved this, since it has gone closed-source.