actuallymentor/battery #382: Battery manager no longer works with macOS 26. Anybody else experiencing this? Any solutions?

wellzhi, 2025-09-16:
Battery manager no longer works with macOS 26. Anybody else experiencing this? Any solutions?

kw-sudo, 2025-09-16:
+1 here

TomatoChat, 2025-09-16:
There's already an issue open for this one (#381)

romu70, 2025-09-16:
+1 (M4 Pro)

luochanghao066, 2025-09-17:
Boss, hurry up and fix it, update and upgrade.

str4ngeMD, 2025-09-19:
> I think this blog, will explain others the reason behind this https://medium.com/@akhshyganesh/why-your-battery-limiter-isnt-working-after-macos-26-update-a-deep-dive-into-the-issue-85493f40f60d

Why the dislikes on this? It provides important context.

iammicloud, 2025-09-21:
> I tested it. It doesn't work. It's not what you need.

I recommend [BatteryOptimizer_for_Mac](https://github.com/js4jiang5/BatteryOptimizer_for_Mac), which is a fork of this battery app, and it offers sail mode for free.