actuallymentor/battery #384: Broken in macOS Sequoia 15.7

natecodes, 2025-09-17:
**What is the issue? (required)**
`battery maintain` no longer works in 15.7
Charge limit is no longer respected

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
Try to use any `battery` feature from CLI or tool as normal.

**Expected behavior (required)**
`battery` should work

**Screenshots (optional)**
n/a

**Error logs**
```
➜  ~ battery logs
👾 Battery CLI logs:

Error: SMCWriteKey() = e00002bc
09/17/25-13:40:15 - MagSafe LED function invoked
09/17/25-13:40:15 - 💡 Setting magsafe color to orange
09/17/25-13:40:15 - setting LED to orange
09/17/25-13:40:15 - MagSafe LED function invoked
09/17/25-13:40:15 - 💡 Setting magsafe color to orange
09/17/25-13:40:15 - setting LED to orange
09/17/25-13:41:15 - Charge below 50
09/17/25-13:41:15 - 🔌🔋 Enabling battery charging
Error: SMCWriteKey() = e00002bc
Error: SMCWriteKey() = e00002bc
09/17/25-13:41:15 - 🔼🪫 Disabling battery discharging
Error: SMCWriteKey() = e00002bc
09/17/25-13:41:15 - Disabling discharging: Charge below 50, enabling charging
Error: SMCWriteKey() = e00002bc
Error: SMCWriteKey() = e00002bc
09/17/25-13:41:15 - MagSafe LED function invoked
09/17/25-13:41:15 - 💡 Setting magsafe color to orange
09/17/25-13:41:15 - setting LED to orange
09/17/25-13:41:15 - MagSafe LED function invoked
09/17/25-13:41:15 - 💡 Setting magsafe color to orange
09/17/25-13:41:15 - setting LED to orange
09/17/25-13:41:23 - 🚨 Calibration process have been stopped
09/17/25-13:41:23 - Not triggering discharge as it is not requested
09/17/25-13:41:23 - Charging to and maintaining at 50% from 47%
09/17/25-13:41:23 - Charge below 50
09/17/25-13:41:23 - 🔌🔋 Enabling battery charging
Error: SMCWriteKey() = e00002bc
Error: SMCWriteKey() = e00002bc
09/17/25-13:41:23 - 🔼🪫 Disabling battery discharging
Error: SMCWriteKey() = e00002bc
09/17/25-13:41:23 - Disabling discharging: Charge below 50, enabling charging
Error: SMCWriteKey() = e00002bc
Error: SMCWriteKey() = e00002bc
09/17/25-13:41:23 - MagSafe LED function invoked
09/17/25-13:41:23 - 💡 Setting magsafe color to orange
09/17/25-13:41:23 - setting LED to orange
09/17/25-13:41:23 - MagSafe LED function invoked
09/17/25-13:41:23 - 💡 Setting magsafe color to orange
09/17/25-13:41:23 - setting LED to orange
09/17/25-13:42:23 - Charge below 50
09/17/25-13:42:23 - 🔌🔋 Enabling battery charging
Error: SMCWriteKey() = e00002bc
Error: SMCWriteKey() = e00002bc
09/17/25-13:42:24 - 🔼🪫 Disabling battery discharging
Error: SMCWriteKey() = e00002bc
09/17/25-13:42:24 - Disabling discharging: Charge below 50, enabling charging
Error: SMCWriteKey() = e00002bc
Error: SMCWriteKey() = e00002bc
09/17/25-13:42:24 - MagSafe LED function invoked
09/17/25-13:42:24 - 💡 Setting magsafe color to orange
[... 148 lines cut ...]
checking-for-update
[ update-electron-app ]
update-not-available

📁 Config folder details:

total 34424
drwxr-xr-x@  8 nathan  staff   256B Sep 17 13:41 .
drwxr-x---+ 68 nathan  staff   2.1K Sep 17 13:58 ..
-rwxr-xr-x@  1 nathan  staff   2.0M Sep 17 13:58 battery.log
-rw-r--r--@  1 nathan  staff     6B Sep 17 13:41 battery.pid
-rw-r--r--@  1 nathan  staff     6B Nov 18  2024 calibrate.pid
-rw-r--r--@  1 nathan  staff    13M Sep 17 13:58 gui.log
-rw-r--r--@  1 nathan  staff     3B Sep 17 13:41 maintain.percentage
-rw-r--r--   1 nathan  staff   965B Jul 11 07:37 visudo.tmp

⚙️	Battery data:

09/17/25-13:58:35 - Battery at 63% (5:20 remaining), 11.8V, smc charging disabled
09/17/25-13:58:35 - Your battery is currently being maintained at 50%
Battery CLI utility v1.2.7
```

**Additional context (optional)**
Add any other context about the problem here.

iammicloud, 2025-09-18:
> So it seems the firmware update is installed in 15.7 and mac os tahoe which breaks the current implementation.
> 
> I believe the tool he uses https://github.com/hholtmann/smcFanControl.git to modify the bits has been broken.
> 
> [charlie0129/batt#34](https://github.com/charlie0129/batt/issues/34) is facing similar issues
> 
> and also, the guy that maintains that library has abandoned it. he is probably going to have to find a new tool or figure out how to change the bits using iokit.pwr_mgt like al dente is doing.
> 
> i hope he manages to do a cli solution as thats the most lightweight.

Not true. [This fork](https://github.com/js4jiang5/BatteryOptimizer_for_Mac) works well and it's a cli only solution.

anarchy89, 2025-09-18:
actually, https://github.com/charlie0129/batt/blob/master/pkg/smc/consts_arm64.go#L10 seems like these are the new smc keys

zaigham, 2025-09-20:
macOS 15.7 (24G222).

same here, battery is no longer cutting off the charge at the set limit.

```
⚙️       Battery data:

09/20/25-19:53:36 - Battery at 89% (0:45 remaining), 12.873V, smc charging disabled
09/20/25-19:53:36 - Your battery is currently being maintained at 80%
Battery CLI utility v1.2.7
```

Mccranky83, 2025-09-20:
Same here. Upgraded to macOS 14.8 and battery just stopped working.

ohtwadi, 2025-09-21:
Just upgraded to 15.7. Took me a bit to notice that battery charge was at 96%. I've changed nothing else.

iammicloud, 2025-09-21:
I recommend [BatteryOptimizer_for_Mac](https://github.com/js4jiang5/BatteryOptimizer_for_Mac), which is a fork of this battery app. It supports Tahoe 26 and offers many other features like sail mode and scheduled auto calibration for free. These features are available only in AlDente Pro version.

iammicloud, 2025-09-24:
> > I recommend [BatteryOptimizer_for_Mac](https://github.com/js4jiang5/BatteryOptimizer_for_Mac), which is a fork of this battery app. It supports Tahoe 26 and offers many other features like sail mode and scheduled auto calibration for free. These features are available only in AlDente Pro version.
> 
> How so? I think I saw the same comment you made in another issue (spam?). The project you reference doesn't have Tahoe/SMC related updates and doesn't have any issue opened at all. So I highly doubt decent user usage/feedback suggesting it's a better alternative than this current project.

Can you read? There're no open issues because they've been solved and closed.

rodalpho, 2025-09-25:
This project is taking too long to update so I switched to the batteryoptimizer which seems to work fine. It is CLI only though; I liked having a little menubar app for this. Not a big deal.

ohtwadi, 2025-09-25:
I am now using the free version of AlDente. The free version is all I need. It has a menubar icon.

NapoelonIV, 2025-09-30:
Has anyone tried battery app in macOS 15.7.1? It just received another update.