actuallymentor/battery #386: Not working on MacOS 26.0

ManisCloud-Yang, 2025-09-22:
Not working on MacOS 26.0

[image]

alvarogarciapiz, 2025-09-22:
Same issue here

toomin5, 2025-09-25:
Same here mac os 26...

deltaSZ3ro, 2025-09-28:
Apple is truly awful. This update signals the end of third-party battery management tools. I can't tell what they intended from a layperson's perspective, but it's truly awful.

ashlord666, 2025-10-02:
I installed Tahoe 26.0.1 and didn't do anything else, and battery started working properly again. Now it says Batt: 80% (on hold) on Apple's own battery icon.

AtaanshAhlawat, 2025-10-02:
> Guys, I found a way. https://github.com/charlie0129/batt Please use it only if you want to.

Thanks bud. This works.

volcbs, 2025-10-06:
@base47 Can confirm that it works now, even when the "optimized battery charging" is still enabled.